erpnet/ErpNet.FP #131: Empty quantity for Datecs X models

iveivanov, 2019-12-16:
We made with 
https://github.com/erpnet/ErpNet.FP/issues/120

to not printing quantity when it is not present in JSON. This is not working for some Datecs models.
Can we add support for Datecs X models? There are additional setings for that.

g-vvv, 2019-12-16:
I think we need device with the new firmware and protocol documentation for it. I will research the case.

g-vvv, 2019-12-16:
As I expected, a new firmware is needed. There will be X models with new firmware, or firmware update at the end of December, or in the beginning of January. After that update we will be able to check the default behaviour of empty "quantity" field in the protocol. I will leave this issue open until then.

g-vvv, 2019-12-20:
It is expected that the default behaviour of the new firmware will accept empty quantity in the same way as in the older protocols. We will reopen this issue if it is not the case, when the new firmware arrives.

closed 2019-12-20 by g-vvv