erpnet/ErpNet.FP #132: Montly Z report and Duplicates

pvelkov, 2019-12-18:
We are integrating the library with our SAP instance now and it runs perfectly. However, we still miss 2 features - printing montly (periodical) Z report and the ability to print duplicate receipts. We are aware that the montly Z report will be deprecated in a couple of months, however we will have a period when we will need this feature.

i-argentinski, 2019-12-18:
For the interim period, you can use:
https://github.com/erpnet/ErpNet.FP/blob/master/PROTOCOL.md#post-post-raw-request

This can be customized to do what you need. But you will have to dig in the protocols of your specific printers.

pvelkov, 2019-12-18:
OK, got it, thx! One additional question regards the raw commands - is cyrillic supported by the web-server? I'm trying to read data from KLEN to visualize the fiscal receipts, but cyrillic is not being displayed correctly. Neither in the log, nor in the response.

synapio, 2019-12-18:
Read about json utf-8 standart.

g-vvv, 2019-12-18:
You are right @pvelkov. The input - raw request, and the result (output) - raw response should be in the same encoding. Now the input is converted from utf-8 to the printer's encoding which is in most cases cp1251, but the result is raw (in cp1251). It can be more consistent if the output is converted from the printer's encoding back to utf-8. I will make this issue as an enhancement, and in the new release, this functionality will be made.

palex, 2019-12-18:
Reports will be dropped from administration requirements, but accountants still needs for its, so on one way or another this feature has to be done in the production system. 
Obviously g-vv created rawrequest as a clever way to get out :)

g-vvv, 2019-12-18:
@palex, you have the freedom to publish your changes and patches here, do not keep them only for you. Share the code, for the community. Or I am the only person who wants to give something away for free :) ?

closed 2019-12-20 by g-vvv