erpnet/ErpNet.FP #133: Status code 405

f1mp3r, 2019-12-23:
Sometimes when requesting different endpoints it returns status code 405 out of nowhere and does this no matter how many times I retry the command. I have to manually restart the server to fix this.  It's done this on different servers connected with FP-700X and DP-150. Version is 1.0.1122.1122

**Log**
```
[2019-12-20 18:34:26 INF] Request starting HTTP/1.1 GET http://192.168.100.35:8001/printers/dt785811/status  
[2019-12-20 18:34:26 INF] -- HTTP Request -- GET: http://192.168.100.35:8001/printers/dt785811/status, Body(0):
[2019-12-20 18:34:26 INF] Executing endpoint 'ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server)'
[2019-12-20 18:34:26 INF] Route matched with {action = "Status", controller = "Printers"}. Executing controller action with signature Microsoft.AspNetCore.Mvc.ActionResult`1[ErpNet.FP.Core.DeviceStatusWithDateTime] Status(System.String) on controller ErpNet.FP.Server.Controllers.PrintersController (ErpNet.FP.Server).
[2019-12-20 18:34:26 INF] Executing ObjectResult, writing value of type 'Microsoft.AspNetCore.Mvc.ProblemDetails'.
[2019-12-20 18:34:26 INF] Executed action ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server) in 102.8292ms
[2019-12-20 18:34:26 INF] Executed endpoint 'ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server)'
[2019-12-20 18:34:26 INF] -- HTTP Response -- Code: 405, Body(54):
{"status":405,"traceId":"|68105e21-439889b41cb1c0ac."}
[2019-12-20 18:34:26 INF] HTTP GET /printers/dt785811/status responded 405 in 175.5647 ms
[2019-12-20 18:34:26 INF] Request finished in 175.9478ms 405 application/problem+json; charset=utf-8
```

g-vvv, 2019-12-23:
Can you reproduce this with the latest version? It looks like Kestrel problem...

f1mp3r, 2019-12-23:
Will update soon and report again if it happens.

g-vvv, 2019-12-23:
Status405MethodNotAllowed, is when the context is not ready to accept your requests. Did you forget your admin page in discovery mode. The context is not ready only when the autodiscovery is running.

f1mp3r, 2019-12-23:
It should be off, will check when I can. Thank you.

closed 2019-12-23 by g-vvv

dsetekh, 2020-01-18:
Can you, please, elaborate on this. 
The proper procedure for production would be to manually add the device then and if device change occur then to re-configure? 
Thank you very much.