erpnet/ErpNet.FP #141: ErpNet.FP cannot recognize device power state

cherry83, 2020-01-22:
If the device is powered on after the start of ErpNet.FP, the server cannot recognize (and connect to) it. 
After the service is restarted, the connection is successful. 
If after that I power off the device, the printer is still in 'Available Printers" and if I try to get the status, the response is:
{
   "deviceDateTime":"0001-01-01T00:00:00",
   "ok":false,
   "messages":[
      {
         "type":"error",
         "code":"E107",
         "text":"no response"
      },
      {
         "type":"info",
         "text":"Error occured while reading current date and time"
      },
      {
         "type":"info",
         "text":"Error occured while reading current status"
      },
      {
         "type":"error",
         "code":"E409",
         "text":"Cannot read current date and time"
      }
  ]
}

Why there are four messages, when the first one ( "no response") is enough in that case.

The device is Datecs DP-150 and is added as "Manually Configured"

g-vvv, 2020-01-22:
Please, be active, make a suggestion, how do you think we should fix this issue?

cherry83, 2020-01-22:
I do not know exactly how the device discovery was implemented, but the idea: when a request is made to a specific device (from the list of manually configured devices), the procedure for detecting devices to be executed, but only for this port.

g-vvv, 2020-01-23:
Every time, before every single command to the device, to run discovery will be overkill. Please, suggest something else, that will not ruin the performance at least 3 times. Users of ErpNet.FP wanted to check the status fast, so I think the easies way will be to restart the discovery manually with the API, when you find that there is a problem with a device. See how it is done by the Detect button in the admin page and implement it in your program.

closed 2020-01-23 by g-vvv