erpnet/ErpNet.FP #148: Global timeout for job

i-argentinski, 2020-02-21:
The global timeout is the timeout, which the client software is willing to totally wait for a job.

The global timeout includes the time for queue waiting, printing and waiting for hardware issues to be resolved (like cover open, out of paper, etc). It starts ticking from the moment the job is received for queuing. When the time has expired, the printing job is cancelled even if it has already started printing.

The client software may be unaware if there are other jobs in the queue, whether there is a physical problem with the printer or just the printing of the document took too much time. The global timeout is the time it is willing to wait, regardless of the reasons.

The global timeout cannot be less than asyncTimeout. 

**Parameter Format**

timeout=_value_
where _value_ can be:
* Without measurement unit - time in milliseconds.
* ___s, ___m, ___ms - time in seconds, minutes or milliseconds.

NOTE: With this issue, please allow asyncTimeout to also specify measurement unit (m, s or ms).

For example:
https://server/receipt?timeout=3m&asyncTimeout=30s

g-vvv, 2020-02-21:
If m, s or ms suffixes are not specified, then by default will be assumed that the parameter is an integer in milliseconds, which will maintain the backward compatibility.

g-vvv, 2020-03-05:
This feature is released in v.1.0.2020.3000.

closed 2020-03-05 by g-vvv