erpnet/ErpNet.FP #149: Problem with reversal receipt

tyrlaka, 2020-04-09:
Hello,
I have problem with printing reversal receipt. 
This is the error that i receive:
```
["receiptNumber"]=>
  string(0) ""
  ["receiptDateTime"]=>
  string(19) "2020-04-09T20:06:46"
  ["receiptAmount"]=>
  float(89)
  ["fiscalMemorySerialNumber"]=>
  string(8) "02576645"
  ["ok"]=>
  bool(false)
  ["messages"]=>
  array(3) {
    [0]=>
    object(stdClass)#8 (2) {
      ["type"]=>
      string(4) "info"
      ["text"]=>
      string(48) "Error occurred while reading last receipt number"
    }
    [1]=>
    object(stdClass)#7 (3) {
      ["type"]=>
      string(5) "error"
      ["code"]=>
      string(4) "E409"
      ["text"]=>
      string(28) "Last receipt number is empty"
    }
    [2]=>
    object(stdClass)#5 (2) {
      ["type"]=>
      string(4) "info"
      ["text"]=>
      string(42) "Error occured while printing receipt items"
    }
  }
}
```

I check few times and i do send a receiptNumber to the fiscal device. I check also the debug.log there also is written that the fiscal device receive the receiptNumber, but not returning such. What could cause this problem?

g-vvv, 2020-04-09:
Please provide debug.log here.

tyrlaka, 2020-04-09:
[debug.log](https://github.com/erpnet/ErpNet.FP/files/4457835/debug.log)
You can check from 2020-04-09 19:57:33

g-vvv, 2020-04-10:
`{
   "uniqueSaleNumber":"DT576645-0001-0000002",
   "operator":"1",
   "operatorPassword":"1",
   "items":[
      {
         "text":"10180 \u0418\u0441\u0442\u043e\u0440\u0438\u0447\u0435\u0441\u043a\u0430 \u0435\u043d\u0446\u0438\u043a\u043b\u043e\u043f\u0435\u0434\u0438\u044f \u0411\u044a\u043b\u0433\u0430\u0440\u0438\u044f",
         "quantity":"1",
         "unitPrice":"35.00",
         "taxGroup":1
      },
      {
         "text":"222635 \u041a\u043e\u043b\u0445\u0438\u0434\u0430. \u0418\u0437\u0431\u0440\u0430\u043d\u0438 \u0441\u0442\u0438\u0445\u043e\u0442\u0432\u043e\u0440\u0435\u043d\u0438\u044f",
         "quantity":"1",
         "unitPrice":"10.00",
         "taxGroup":1
      }
   ],
   "payments":[
      {
         "amount":45,
         "paymentType":"cash"
      }
   ],
   "receiptNumber":"0001670",
   "receiptDateTime":"2020-04-09 19:45:15",
   "fiscalMemorySerialNumber":"2576645",
   "reason":"refund"
}`
**receiptDateTime** is not well formatted, see the examples for Date and Time formatting.
**unitPrice** is not well formatted, see the examples, avoid using double quotes for numbers.
**quantity** is not well formatted,  see the examples, avoid using double quotes for numbers.

closed 2020-04-10 by g-vvv