erpnet/ErpNet.FP #150: ErpNet.FP return Timeout when the Status command is executed

iveivanov, 2020-06-08:
After print receipt when try to get Status of printer I receive timeout error:

[2020-05-20 14:53:31 INF] Request starting HTTP/1.1 GET http://localhost:8001/printers/dt525860/status  
[2020-05-20 14:53:31 INF] -- HTTP Request -- GET: http://localhost:8001/printers/dt525860/status, Body(0):
[2020-05-20 14:53:31 INF] Executing endpoint 'ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server)'
[2020-05-20 14:53:31 INF] Route matched with {action = "Status", controller = "Printers"}. Executing controller action with signature Microsoft.AspNetCore.Mvc.ActionResult`1[ErpNet.FP.Core.DeviceStatusWithDateTime] Status(System.String) on controller ErpNet.FP.Server.Controllers.PrintersController (ErpNet.FP.Server).
[2020-05-20 14:53:31 INF] Executing ObjectResult, writing value of type 'ErpNet.FP.Core.DeviceStatusWithDateTime'.
[2020-05-20 14:53:31 INF] Executed action ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server) in 3.9626ms
[2020-05-20 14:53:31 INF] Executed endpoint 'ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server)'
[2020-05-20 14:53:31 INF] -- HTTP Response -- Code: 200, Body(368):
{"deviceDateTime":"0001-01-01T00:00:00","ok":false,"messages":[{"type":"error","code":"E999","text":"User timeout occured while sending the request"},{"type":"info","text":"Error occured while reading current date and time"},{"type":"info","text":"Error occured while reading current status"},{"type":"error","code":"E409","text":"Cannot read current date and time"}]}
[2020-05-20 14:53:31 INF] HTTP GET /printers/dt525860/status responded 200 in 7.2216 ms
[2020-05-20 14:53:31 INF] Request finished in 9.1841ms 200 application/json; charset=utf-8

There are no communications with the printer in this period - ErpNet.FP return this error without check printer.

We use asyncTimeout for requests. For example:

[2020-06-01 16:44:05 INF] Request starting HTTP/1.1 POST http://localhost:8001/printers/dt525860/receipt?timeout=180000&asyncTimeout=0&taskId=f30eea1953d543e1a9a3d19096549363 application/json 1076
[2020-06-01 16:44:05 INF] -- HTTP Request -- POST: http://localhost:8001/printers/dt525860/receipt?timeout=180000&asyncTimeout=0&taskId=f30eea1953d543e1a9a3d19096549363, Body(1001):
{"uniqueSaleNumber":"DT525860-0001-0000134","items":[{"text":"Оператор: 0001","type":"comment"},{"text":" ","type":"comment"},{"text":"Артикул          Кол.   Ед.цена           Общо","type":"comment"},{"text":"----------------------------------------------","type":"comment"},{"text":"                 2.00 x 180.0000        360.00","type":"comment"},{"text":"Гел","unitPrice":360.000,"taxGroup":2},{"text":"----------------------------------------------","type":"comment"},{"text":"                 1.00 x  65.1600         65.16","type":"comment"},{"text":"Компютърна конфигурация","unitPrice":65.160,"taxGroup":2},{"text":"----------------------------------------------","type":"comment"},{"text":"                 1.00 x   0.7100          0.71","type":"comment"},{"text":"Компютърна конфигурация","unitPrice":0.710,"taxGroup":2},{"text":"----------------------------------------------","type":"comment"}],"payments":[{"paymentType":"cash","amount":175.870},{"paymentType":"card","amount":250.000}]}

I make tests on version 1.0.2020.3000

iveivanov, 2020-06-08:
The printer is FP-700X

iveivanov, 2020-06-08:
At the same time Printer info command return correct result:
[image]

g-vvv, 2020-06-08:
This bug is reproduced and will be fixed in the next release. Thank you for the feedback.

closed 2020-06-27 by g-vvv