erpnet/ErpNet.FP #154: Discount calculation on negative valued item

iarn-sl, 2020-06-30:
Issue:
When an amount discount is applied to an negative price item the calculation is incorrect.

Example :
```
    {"text":"Test SKU","quantity":"1.000","unitPrice":"-50.00","taxGroup":"2",
    "priceModifierValue":"15",
    "priceModifierType":"discount-amount"}
```

Calculated as : 65 instead of 35

Additional Information:
Percent discount ( discount-percent ) is correctly calculated and the expect behavior is observed when used.

Version tested:
 v.1.0.2020.3000

Reproduce :

```
{"uniqueSaleNumber":"DT800001-0001-0000001","items":[
    {"text":"SKU_1","quantity":"1",
    "unitPrice":"40.00","taxGroup":"2",
    "priceModifierValue":"5.00",
    "priceModifierType":"discount-amount"},
    {"text":"SKU_2","quantity":"1.000","unitPrice":"40.00","taxGroup":"2",
    "priceModifierValue":"5.00",
    "priceModifierType":"discount-amount"},
    {"text":"Test SKU","quantity":"1.000","unitPrice":"-50.00","taxGroup":"2",
    "priceModifierValue":"15",
    "priceModifierType":"discount-amount"}]
,"payments":[
{"amount":"40.00","paymentType":"cash"},
{"amount":"-5.00","paymentType":"change"}
]}
```

Debug excerpt :

> [11:59:09 INF] -- HTTP Request -- POST: /receipt, Body(612):
> {"uniqueSaleNumber":"DT880001-0001-0000001","items":[
>     {"text":"SKU_1","quantity":"1",
>     "unitPrice":"40.00","taxGroup":"2",
>     "priceModifierValue":"5.00",
>     "priceModifierType":"discount-amount"},
>     {"text":"SKU_2","quantity":"1.000","unitPrice":"40.00","taxGroup":"2",
>     "priceModifierValue":"5.00",
>     "priceModifierType":"discount-amount"},
>     {"text":"Test SKU","quantity":"1.000","unitPrice":"-50.00","taxGroup":"2",
>     "priceModifierValue":"15",
>     "priceModifierType":"discount-amount"}]
> ,"payments":[
> {"amount":"40.00","paymentType":"cash"},
> {"amount":"-5.00","paymentType":"change"}
> ]}
> [11:59:09 INF] Executing endpoint 'ErpNet.FP.Server.Controllers.PrintersController.PrintReceipt (ErpNet.FP.Server)'
> [11:59:09 INF] Route matched with {action = "PrintReceipt", controller = "Printers"}. Executing controller action with signature System.Threading.Tasks.Task`1[Microsoft.AspNetCore.Mvc.IActionResult] PrintReceipt(System.String, ErpNet.FP.Core.Receipt, System.String, System.String, Int32) on controller ErpNet.FP.Server.Controllers.PrintersController (ErpNet.FP.Server).
> [11:59:09 INF] Executing ObjectResult, writing value of type 'ErpNet.FP.Core.DeviceStatus'.
> [11:59:09 INF] Executed action ErpNet.FP.Server.Controllers.PrintersController.PrintReceipt (ErpNet.FP.Server) in 7.2407ms
> [11:59:09 INF] Executed endpoint 'ErpNet.FP.Server.Controllers.PrintersController.PrintReceipt (ErpNet.FP.Server)'
> [11:59:09 INF] -- HTTP Response -- Code: 200, Body(146):
> {"ok":false,"messages":[{"type":"error","code":"E403","text":"Payment total amount (35.00) should be the same as the items total amount (5.00)"}]}
> [11:59:09 INF] HTTP POST /printers/dt000000/receipt responded 200 in 24.8730 ms

g-vvv, 2020-06-30:
I think that a discount applied on a negative unit price is not something that one would actually use. But yes, it is an interesting finding, so please feel free to offer a patch/Pull Request for the fiscal receipt's validation function, and I will be glad to apply it to the master branch.  See https://github.com/erpnet/ErpNet.FP/blob/master/ErpNet.FP.Core/Drivers/BgFiscalPrinter.cs, method ValidateReceipt.
Thank you.

closed 2020-06-30 by g-vvv

wqweto, 2020-06-30:
Perhaps a simple `Math.Sign(itemPrice)` can be used like this

    case PriceModifierType.DiscountAmount:
        itemPrice -= Math.Sign(itemPrice) * itemPriceModifierValue;

. . . and in surcharge case as well

    case PriceModifierType.SurchargeAmount:
        itemPrice += Math.Sign(itemPrice) * itemPriceModifierValue;

(not tested)