erpnet/ErpNet.FP #157: Datecs WP-500 support protocol

dsphinx, 2020-12-19:
i cant find my device Datecs WP-500 ( via TCP ),
i tried all the bg.dt.xx.xx protocols but without success .
Can you help me ?

wqweto, 2020-12-20:
> i tried all the bg.dt.xx.xx protocols but without success .

Can you post some printer Uri that failed?

Try something like `bg.dt.x.isl.tcp://192.168.1.77:4999` for printer Uri (but adjust IP beforehand). This uses Datecs X models protocol on default 4999/tcp port.

Btw, TCP/IP connectivity has to be enabled manually from ECR settings.

iveivanov, 2020-12-20:
Hello,

wqweto, 2020-12-20:
Facebook groups do not support markdown and the occasional ErpNet.FP threads get lost in BgBusinessDev group's history.

There are Discussions available in repo's settings on github now. Might be helpful to cleanup issues from support requests.

dsphinx, 2020-12-21:
Tried Uri:
i ) bg.dt.x.isl.tcp://192.168.54.151:9001
ii) 192.168.54.151:9001

The printer is up and running in LAN  - ping 0% packer loss
PING xxx (192.168.54.151) 56(84) bytes of data.
64 bytes from xxx (192.168.54.151): icmp_seq=1 ttl=64 time=1.19 ms

[2020-12-21 09:13:37 INF] Trying Datecs wp-500: bg.dt.x.isl.tcp://192.168.54.151:9001, failed

LOGS from Server: 
[2020-12-21 09:11:21 INF] Trying Datecs wp-500: 192.168.54.151:9001, failed
[2020-12-21 09:11:21 INF] -- HTTP Response -- Code: 200, Body(68):
{"Datecs wp-500":{"id":"Datecs wp-500","uri":"192.168.54.151:9001"}}

i-argentinski, 2020-12-21:
The primary way ErpNet.FP was designed to be used is as a network print server, sitting NEAR the printer (in network aspect). By using the TCP protocol of the printer, you are waiving the network proficiency of erpnet.fp. You start to rely solely on the network abilities of the printers, which are not very high.

The best practice is to have 1 erpnet.fp server for each PC, which has fiscal printer(s). Then, employ the network protocol and queue of erpnet.fp to establish a stable communications environment.

dsphinx, 2020-12-21:
i just get responce from server ,
{"status":405,"traceId":"|6da9ad93-41e877c460978f47."}

wqweto, 2020-12-21:
> Tried Uri:
> i ) bg.dt.x.isl.tcp://192.168.54.151:9001
> ii) 192.168.54.151:9001

The port looks wrong because the default port for Datecs X models is 4999.

Try `bg.dt.x.isl.tcp://192.168.54.151:4999`

dsphinx, 2020-12-21:
i tried that tcp port 4999 but it isn't open,
the only open tcp portw on the fiscal printer are  tcp 9001 and tcp 9100 .

I am from Greece, and maybe tha firmware was altered from the GR reseller - solidus - that why the changed port -,
but from other software tha worked with the printer says that the protocol is SOLIDUS (II) T. obviously the change the tcp port but the main functions must be the default bg.dt.x.isl.

wqweto, 2020-12-21:
Port 9100/tcp is used by non-fiscal ESC/POS "kitchen receipt" printers.

Do you use fiscal receipts in Greece at all? I doubt it as Google search shows no relevant info.