erpnet/ErpNet.FP #159: Using bg.dt.x.isl driver with DP-25 device

rcturk, 2021-02-08:
First of all, thank you for this repository. 

I am trying to connect a DP-25 device with ErpNet.FP. The problem is, bg.dt.c.isl driver is not working, but bg.dt.x.isl is.

Device i tried to connect is from Romania. Maybe configurations differ between countries. 

First difference is serial number length is 12 on this device.
Other one is device info command does not return fiscalMemorySerialNumber.

Here is a screenshot.

[image]

Here is device info result from http://localhost:8001/printers/db4200076708:

`{"uri":"bg.dt.x.isl.com://COM1","serialNumber":"DB4200076708","fiscalMemorySerialNumber":"","manufacturer":"Datecs","model":"DP-25","firmwareVersion":"410236 14Feb18 1000","itemTextMaxLength":72,"commentTextMaxLength":40,"operatorPasswordMaxLength":8,"taxIdentificationNumber":"","supportedPaymentTypes":["cash","check","coupons","ext-coupons","card"],"supportsSubTotalAmountModifiers":true}`

I wanted to ask if you have any issue like this before.

Thank you in advance.

clauub, 2022-04-29:
Hi, @ridvandev 
I am trying to make this repository to work with fiscal printer DP-25 in Romania. But it's not working with any of the protocols. 

How did you fix this to work, did you follow any steps when starting and configuring the printer with the server? 

Thank you!

ridvandev, 2022-04-29:
Hi, @clauub,
You may look at the fork: https://github.com/ridvandev/ErpNet.FP
I have a commit to fix some commands: [2ed544b96237260c0c068e19416488e95c1deb68](https://github.com/ridvandev/ErpNet.FP/commit/2ed544b96237260c0c068e19416488e95c1deb68)

clauub, 2022-05-05:
@ridvandev thank you, it's working