erpnet/ErpNet.FP #163: DATECS PAYMENT TYPE DEBIT CARD DOES NOT EXIST?

krcenov, 2021-06-22:
I have a Datecs WP-50X and i want it to make a sale with payment type 2
these are the payments in the device
[image]
i can use all other payment types, but debit card is missing.
[image]
cash-0
check-3
coupons-5
ext-coupons-4
card-1
PAYMENT 2 IS MISSING

iveivanov, 2021-06-22:
Why don't you use Card - 1 type? NRA don't distinguish credit or debit card nor sales person distinguish them when accept card payment.

krcenov, 2021-06-22:
> Why don't you use Card - 1 type? NRA don't distinguish credit or debit card nor sales person distinguish them when accept card payment.

Beacause it does not trigger POS terminal payment.
[image]

g-vvv, 2021-06-22:
**POS terminal integration is not implemented in ErpNet.FP**. So, even if this type of payment (debit card - 2) is appended to the list of supported payment types, **the instruction for initialization of the communication with the POS terminal will not be sent**. The problem with the POS terminals must be solved by implementing that support in all FP devices, by their custom protocol requirements. **Feel free to implement this functionality, it would be useful to many POS/ErpNet.FP users.**

krcenov, 2021-06-22:
I have implemented it half way for the X devices in the Datecs protocol, since i need this for customers.
I am a DATECS employee from the Software department.

krcenov, 2021-06-22:
> **POS terminal integration is not implemented in ErpNet.FP**. So, even if this type of payment (debit card - 2) is appended to the list of supported payment types, **the instruction for [...]

Your statement is partialy correct.
Using payment type 2 does initialise communication with the POS terminal.
And the terminal displays the sum needed to pay.
When you pay the sum, the WP-50X then prints the reciept.
The problem is that the ERP does not send command 55,15\t in order to print the card transaction details.

g-vvv, 2021-06-22:
Yes this is what I meant: "The problem is that the ERP (ErpNet.FP) does not send command 55,15\t in order to print the card transaction details." I do not have POS terminal device with me to test that functionality, so feel free to implement and test it. Then do a pull request with your implementation, and we will be happy to merge it with the master branch. Of course **the presence or absence of the POS terminal must be gracefully handled by your implementation**, otherwise we will break the compatibility with the current version protocol and usage patterns.

wqweto, 2021-06-22:
Btw, for X models sending `55 (37h) Pinpad commands` w parameter `15\t` after each and every `56 (38h) Close fiscal receipt` just returns a harmless `ERR_PINPAD_NOT_CONF` w/ no side-effects as far as I can see.

Anyway, I'll wait for proper Datecs PR to be sure this extra command would be enough to handle pinpads receipts and for the fix to get tested w/ real pinpads (by Datecs).

krcenov, 2021-06-22:
> Yes this is what I meant: "The problem is that the ERP (ErpNet.FP) does not send command 55,15\t in order to print the card transaction details." I do not have POS terminal device with me to test [...]

A colleage of mine representative of DatecsPay should get in touch with you soon.

i-argentinski, 2021-07-19:
We are studying the possible solutions. One possible solution is to implement payment terminal activation TRANSPARENTLY to the user application.

### TRANSPARENT to the user app

1. There will be a setting in ErpNet.FP admin interface, called "Automatically Activate Payment Terminal" (hint "For fiscal notes with CARD payment, activate the payment terminal attached to the fiscal printer to receive the payment.").
2. This setting will appear only for protocols, which support attached payment terminals (currently, it will be only Datecs X).
3. When the setting is activated, FP will dispatch the necessary commands to activate the payment terminal when a note with payment = CARD is being printed.
4. If the payment is unsuccessful, the whole fiscal note will be voided.
5. Possibly, in the return status message to the user app we might include information that the payment was processed by the payment terminal.

### EXPLICIT control of the user app

If we want to provide explicit control to the user app, this will require changes in the protocol which will make the protocol more command-oriented (containing commands what to do) and not so declarative (containing just a description of the fiscal operation). This will minimize the usefulness of the protocol for generic information transfer and will focus it too much on just the task of printing to fiscal printers. We would like to avoid this, if possible. In the future, if there is some form of electronic fiscal data transfer to the government and the protocol should be up to the task (if possible). For that, we shall keep it declarative and simple as much as possible.

Please comment on the proposal.

bentoart, 2023-10-25:
Hello, 
we have little problem with devices with DATECS PAY & WP-150 MX fiscal device, when user push end of the working day "Z-report" we don't have POS report for the day and we must to disconnect the fiscal device from PC and then to hit manually POS report for the day, because if we don't do that, user can't work again with POS on the next working day.

o-ivanov, 2023-10-30:
This is the last version:
https://github.com/erpnet/ErpNet.FP/releases/tag/v1.1.0.721

The only condition to make report from the payment terminal together with the main report of the device is that there is a connection (active and connected bluetooth) between them at the time of launching the report.
Both types of reports (Z and X) are supported.
Support is from this version: https://github.com/erpnet/ErpNet.FP/releases/tag/v1.1.0.318

bentoart, 2023-12-30:
Hello, again, 
can you please tell me which model of the Datecs fiscal devices and datecs pay devices we to use, to have optimal work with Erp.NET.FP, please make me a recommendation.

o-ivanov, 2024-01-03:
I'm sorry, but we cannot provide a recommendation as we lack experience in practical work with cash registers. Our business is not related to this. Please contact DATECS support for assistance.