erpnet/ErpNet.FP #166: Issues with Eltrade A1

skito, 2021-09-01:
Hi Guys,

Is there anything special you need to configure with Eltrade A1? 

I'm successfully printing on Datecs DP-25, but when using the same request for Eltrade A1, I receive:

`E199 Device general error` followed by `E599 NRA link general error`.

* Operator defaults are 1 / no password. However there is no difference if I provide wrong operator or not. 
* The device is production one and working well in manual mode - connected with the NRA device successfully.
* The device is discoverable
* I'm able to print Z report from http://localhost:8001/

Full request/response log: [log.txt](https://github.com/erpnet/ErpNet.FP/files/7090023/log.txt)

Any thoughts?

iveivanov, 2021-09-01:
Hello,
Please check the Bulgarian fiscal printer setup:
https://github.com/erpnet/ErpNet.FP/blob/master/BGFiscalPrinterSetup.md

skito, 2021-09-01:
@iveivanov 

Did check it now. The BaudRate is 115200 and the device is in mode „1.Регистрация“ selected. 
Any other suggestions?

g-vvv, 2021-09-03:
Please read your error codes and messages: "type":"error","code":"E599","text":"The internal terminal is not working"...
So, you do not have working NRA Terminal. Contact your Eltrade service.

closed 2021-09-03 by g-vvv