erpnet/ErpNet.FP #168: Need Documentation Update to include Datecs Terminal use and examples with Datecs X devices

krcenov, 2021-10-13:
Hello,
Can you please update the documentation, so it describes how to use ErpNet.FP with Datecs X devices connected to a 
Bank Terminal?

o-ivanov, 2021-10-13:
Field "card" in "Payment Type to the printer...." must be blank.
Please check again with blank field.

o-ivanov, 2021-10-13:
Maybe Server is installed in folder where no Write permissions and therefore no way to save/change configuration.

o-ivanov, 2021-10-13:
Is Server running as a Service?
Stop service and start server like application. (in application folder start  ErpNet.FP.Server.exe)

krcenov, 2021-10-13:
After stopping the service and manually starting the Server as administrator, this is my settings json
[image]

This is what is displayed on the UI
[image]
[image]

krcenov, 2021-10-13:
Can i install in C:\New Folder

o-ivanov, 2021-10-13:
Yes

o-ivanov, 2021-10-13:
OK.

krcenov, 2021-10-13:
Using this i installed ERPNET.FP in 
[image]
C:\New folder
My settings is still empty and the checkbox is still missing.