erpnet/ErpNet.FP #181: Has the latest version been tested with Raspberry Pi?

rabchev, 2023-04-07:
I can't seem to be able to run binaries directly. I get the following error when I try to execute ErpNet.FP.Server: "Could not execute because the specified command or file was not found."

However, if I build the binaries on the Raspberry I'm able to start the server, but than apparently there is some thread locking issue. Detecting printers never completes.

wout-callista, 2024-10-21:
@rabchev Were you able to fix this? I'm having the same thread locking issue, detecting never completes.

rabchev, 2024-10-23:
Which Raspberry version are you using? It works fine on newer versions. I don't remember exactly but I think was testing with Raspberry Pi 3 and it didn't work, but it worked with 4 and 5.

wout-callista, 2024-10-24:
> Which Raspberry version are you using? It works fine on newer versions. I don't remember exactly but I think I was testing with Raspberry Pi 3 and it didn't work, but it worked with 4 and 5. Apparently, this is an issue with the .Net framework it self.

I'm using RP4 with  the latest Raspbian OS. When I run the ErpNet.FP.Server it starts up, but when I go to the localhost:8001 it says "Cannot get printers properties" and keeps detecting infinitely.

I've the installfppi.sh script to install everything as well and run the service but then I can't access the localhost page so it seems that doesn't work.