erpnet/ErpNet.FP #182: datecs DT-150 MX  error E409 Last receipt number is empty

TaHuC, 2023-04-19:
The answer from the cash register:
{"receiptNumber":"","receiptDateTime":"2023-04-19T10:36:51","receiptAmount":42.00,"fiscalMemorySerialNumber":"02944406","ok":false,"messages":[{"type":"info","text":"Error occurred while reading last receipt number"},{"type":"error","code":"E409","text":"Last receipt number is empty"},{"type":"info","text":"Error occured while printing receipt items"}]}

I send: 
{"uniqueSaleNumber":"DT944406-0003-0000004","items":[{"type":"comment","text":" "},{"type":"comment","text":"Оператор: Стамен"},{"type":"comment","text":"Маса: 1"},{"type":"comment","text":" "},{"text":"СМЕТАНА","quantity":1,"unitPrice":0.4,"taxGroup":4}],"payments":[{"paymentType":"cash","amount":0.4}]}

Pls help!!!

iveivanov, 2023-04-19:
Hello, 
Check the operator password in the fiscal device if it matches the default password in ErpNet.FP:
https://github.com/erpnet/ErpNet.FP#default-passwords-we-use-in-the-library

o-ivanov, 2023-04-20:
Hello,
This error arises after sending "close receipt" command. Command fails if total payment amount is less than receipt amount. Fiscal receipt stays open, and neither be finished nor other receipt can be opened.
In your response there is a discrepancy shown between 
    "receiptAmount":42.00 
and
     "paymentType":"cash","amount":0.4

Better use software tool provided by manufacturer of fiscal device to close currently open receipt!

moynzzz, 2023-08-05:
Hello,

I also have the same error on DT-150MX:

Here is my request/response, my pricing is allright:

Request:
`{"uniqueSaleNumber":"DT949353-0001-0000605","items":[{"type":"comment","text":"\u0420\u0435\u0437\u0435\u0440\u0432\u0430\u0446\u0438\u044f #441968"},{"text":"\u041a\u043e\u043d\u0441\u0443\u043b\u0442\u0430\u0446\u0438\u044f \u0441 \u043b\u0435\u043a\u0430\u0440","quantity":1.0,"unitPrice":1.0,"taxGroup":1}],"payments":[{"paymentType":"cash","amount":1.0}]}`

Response:
`{"receiptNumber":"","receiptDateTime":"2023-08-05T12:50:41","receiptAmount":2.00,"fiscalMemorySerialNumber":"02949353","ok":false,"messages":[{"type":"info","text":"Error occurred while reading last receipt number"},{"type":"error","code":"E409","text":"Last receipt number is empty"},{"type":"info","text":"Error occured while printing receipt items"}]}`

rosen4o, 2024-01-30:
I have DP-150MX which is working normally without any problem.
You should change default operator password for operator 1 from 1 to 0000 so everything will work normally without sending operator password.
Just bg.dt.x.isl is using as default operator password for operator 1 - 0000.