erpnet/ErpNet.FP #191: Unable to detect/connect to Datecs DP-150 via TCP

GladBarsan, 2024-12-15:
I am running ErpNet.FP with the following configuration:
```
{
  "ErpNet.FP": {
    "AutoDetect": false,
    "ServerId": "Q4Sn06or1Ea1NBcLCcaj-A",
    "Printers": {
      "DATECS": {
        "Uri": "bg.dt.c.isl.tcp://192.168.1.99:3999"
      }
    },
    "UdpBeaconPort": 8001,
    "PrintersProperties": {}
  },
  "Kestrel": {
    "EndPoints": {
      "Http": {
        "Url": "http://0.0.0.0:8001"
      }
    }
  }
}
```
When starting the service as administrator, it attempts to connect to the specified Datecs DP-150 printer at 192.168.1.99:3999 but fails. The log output repeatedly shows attempts and responses, but ultimately the printer is not detected:
```
[ 2024-12-15 15:13:32 INF] Starting the service, version 1.1.0.1414...
[ 2024-12-15 15:13:32 INF] Detecting configured printers...
[ 2024-12-15 15:13:32 INF] Keep Alive Background Service is started.
[ 2024-12-15 15:13:32 INF] 192.168.1.99:3999 <<< 01-25-21-5A-31-05-30-30-3D-36-03
[ 2024-12-15 15:13:33 INF] 192.168.1.99:3999 >>> 16
[ 2024-12-15 15:13:33 INF] 192.168.1.99:3999 >>> 15
[ 2024-12-15 15:13:33 INF] 192.168.1.99:3999 <<< 01-25-21-5A-31-05-30-30-3D-36-03
... (repeated attempts)
[ 2024-12-15 15:13:33 INF] Trying DATECS: bg.dt.c.isl.tcp://192.168.1.99:3999, failed
[ 2024-12-15 15:13:33 INF] Detecting done. Found 0 available printer(s).
```

While ErpNet.FP is running and making these attempts, the fiscal printer becomes inaccessible to other tools—indicating that ErpNet.FP is successfully locking or occupying the connection. Once I stop ErpNet.FP, I can immediately connect to the printer using the official Datecs utility "DUDE" (from the same computer, same IP/port).

This suggests that the printer is online and reachable, and that the IP/port configuration is correct. The issue seems to be that ErpNet.FP cannot properly handshake or finalize the connection, resulting in a failure to detect the printer.

**Expected Behavior:**
ErpNet.FP should successfully detect and connect to the configured Datecs DP-150 printer, making it available for fiscal operations.

**Actual Behavior:**
ErpNet.FP attempts a connection multiple times and then reports failure, leaving the printer undetected. While this is happening, other applications cannot connect, implying that ErpNet.FP is holding the connection but not completing it successfully.

Steps to Reproduce:
1. Configure ErpNet.FP with the provided settings pointing to a Datecs DP-150 printer, with _bg.dt.c.isl.tcp_ protocol
2. Start ErpNet.FP as administrator.
3. Observe the repeated attempts and eventual failure in the logs.
4. Note that during operation, the printer is not accessible by other tools.
5. Stop ErpNet.FP and attempt to connect via Datecs DUDE—this works as expected.

Environment:
- ErpNet.FP version: 1.1.0.1414
- Printer model: Datecs DP-150
- OS: Windows 11
- Network: Printer accessible at 192.168.1.99:3999

iveivanov, 2024-12-16:
But Datecs DP-150 is cash register and don't have LAN port. 
The protocol bg.dt.c.isl.tcp is for fiscal printer not for cash registers

https://www.datecs.bg/en/products/287

iveivanov, 2024-12-16:
Try using the other TCP protocols of Datesc:

bg.dt.p.isl.tcp and bg.dt.x.isl.tcp